esp32-si4732/ats-mini #1: Volume - Long Press

G8PTN, 2025-03-12:
Hi Max,
I have only had a brief check of behaviour with v1.04.

1. Might be a good idea to do the audio unmute on volume long press. At the moment, if mute is active, the status shows "Muted" even after adjustment of volume.

2. Probably need to add workaround for the auto capacitor behaviour for LW/MW to avoid confusion with RSSI level change.

3. Users may also want the compiled binary files for direct flash download.

Great work, well done.

73, Dave

max-arnold, 2025-03-12:
Thanks for the feedback, Dave!

1. Good idea, will implement
2. Will experiment later
3. This is what I'm focused on right now

max-arnold, 2025-03-12:
Reopening this as a reminder for myself

max-arnold, 2025-03-16:
Number 3 is done, see the [Releases](https://github.com/esp32-si4732/ats-mini/releases) section

max-arnold, 2025-03-20:
Item 1 was released in v1.06

max-arnold, 2025-03-22:
Item 2 (the tuning capacitor) was fixed in #13

closed 2025-03-22 by max-arnold