esp32-si4732/ats-mini #3: Documentation

max-arnold, 2025-03-13:
Convert Dave's documents and notes to Markdown and publish to GH Pages

G8PTN, 2025-03-13:
Hi Max,

- QA and Feature Requests Log
For completeness, I have updated the spreadsheet. You are welcome to use the information, and I do not plan to update it on my GitHub page. Please see attached. 

[ATS_MINI_REQ_QA_3.zip](https://github.com/user-attachments/files/19225878/ATS_MINI_REQ_QA_3.zip)

- General
There are probably many spelling (typo) errors in the code and documentation, I know you have corrected several. I did notice that the comment for the pinout table is wrong. It should read “The relevant columns”.

Do you want me to put a reference on my GitHub pointing to your site?

73, Dave

max-arnold, 2025-04-28:
The documentation is mostly done (it finally has a User Manual!): https://esp32-si4732.github.io/ats-mini/

closed 2025-04-28 by max-arnold