esp32-si4732/ats-mini #5: Seek up/down step on AM/FM

max-arnold, 2025-03-13:
https://github.com/G8PTN/ATS_MINI/issues/11

nicenemo, 2025-04-20:
An easy to use sticky seek mode would be nice. 
The mentioned implementation does work but the UX could be better.

As a user I want:
1. after entering seek-mode turn left or right is seek down/up
2. press to exit
3. exit after a longer timeout when not seeking

max-arnold, 2025-04-20:
This issue is about a different thing - the actual step that is used in the automatic seek mode (as opposed to the manual tuning step).

With that said, a UX similar to what you suggested is implemented in the Chinese v2 firmware: https://github.com/zhang-chong/4732mini/

More options are considered in the following discussion, feel free to add you scenario to it: https://github.com/esp32-si4732/ats-mini/discussions/26

nicenemo, 2025-04-20:
That discussion is fine without me adding noise. It looks a bit like macros for ZMK or QMK mechanical keyboards.  I am sure you come up with something that works well.

max-arnold, 2025-05-13:
* Since version 2.13 seek timeout is 10 minutes instead of 8 seconds
* Seek mode (a menu option) was released in 2.13
* Press to exit will be released in 2.15
* AM/FM seek spacing that corresponds to the current tuning step will be released in 2.15

closed 2025-05-13 by max-arnold