esp32-si4732/ats-mini #9: FM stereo and RDS?

terrancewong, 2025-03-17:
Is it possible to turn them on?
My radio comes with RDS and stereo, lost them after flashing ats-mini.

max-arnold, 2025-03-19:
Yes, it is possible. The tricky part is where to fit it on the screen without cluttering the UI

stevecrow74, 2025-03-19:
One option is have a menu item called RDS info, when clicked it pops up in the menu screen. 
Another option is decrease the current RDS by 25% 
And fit it in below in smaller text.

max-arnold, 2025-03-29:
Thanks for the feedback!

I do plan to expand the RDS info. But for now I'm hesitant to add UTC time (other than just displaying it as part of the RDS data), because it is a can of worms:

1. Some FM stations report wrong time
2. There is no way to keep the clock time ticking when the receiver is powered off, so if you listen for non-FM bands or there is no RDS, the time will be incorrect
3. There is a firmware version that uses NTP over WiFi to sync the time https://github.com/superogira/ATS_MINI but WiFi is power hungry and likely adds more QRM

The stereo indicator is a personal preference (it was an attempt to declutter the UI), I might change it in future. In the meantime, you can try another opinionated fork: https://github.com/m2608/ats_mini

The fonts are really cool! Unfortunately, they will require fiddling with the text positioning and will make the color theme structure more complex. I believe stuff like this is best implemented as a personal fork. This repo has an automated build pipeline, so for simple tweaks you can fork it, edit the files using just the browser, and the rest will be handled for you. Here is a short demo video https://t.me/talkradio/174172/177902 (requires a Telegram account, if you are familiar with GitHub there will be nothing new for you in the video)

Sebastsch, 2025-03-29:
Thank you very much for your responsiveness. Thank you for the Telegram video, I didn't know about it. Thank you and good luck with your development.

andimik, 2025-04-05:
Any chance to send raw RDS via serial port?

Currently I just get the frequency out.

`109,10690,0,VHF,0,0,0,4,4,22,63,0,1,2554,41`

And regarding Wifi: It's really less noise than via serial port (tested with TEF6686).

luarvique, 2025-04-22:
Expanded RDS info options are now available from the main branch. These include:

-- PI Code
-- Station Name
-- Radio Text + Program Type
-- Time

See "Settings | RDS" for available combinations.

luarvique, 2025-04-28:
Is this issue still relevant, given that RDS has been fully implemented?

max-arnold, 2025-04-29:
The RDS was released in 2.12, I think we can close the issue.

closed 2025-04-29 by max-arnold