esp32-si4732/ats-mini #13: Antenna Tuning Capacitor

G8PTN, 2025-03-20:
Hi Max,
When I checked the PU2CLR code for getAntennaTuningCapacitor() function, it does not force a read of the values from the SI4732/5, it uses the value that was read back following for example getFrequency().

At the moment if you connect a whip antenna, the debug will show the following from boot.
VHF: Capacitor value = 0
VHF > MW1: Capacitor value = 0
MW1 > MW2: Capacitor value = 0
Change frequency: Capacitor value = 6143
Screen dump below.
[image]

It may be better to add rx.getFrequency() before the getAntennaTuningCapacitor() as shown below.

// Use rx.getFrquency to force read of capacitor value from SI4732/5
rx.getFrequency();
uint16_t tuning_capacitor = rx.getAntennaTuningCapacitor();

With the change and the same setup, the screen dump is shown below.
[image]

If you then perform the same test with the supplied loop antenna, you will find that the value will not be at maximum, and it will change with frequency as it finds the resonant tuning point when operating in automatic mode.

As stated previously, the calls for rx.setTuneFrequencyAntennaCapacitor(0); are not positioned in the correct place in the code and the automatic mode is only beneficial with a loop antenna.

If you comment out the calls rx.setTuneFrequencyAntennaCapacitor(0); the capacitor value should always be equal to 1 (non-automatic, minimum capacitance). Whilst this is not the ideal point for a loop antenna, it is probably more suitable for the ATS-Mini hardware setup as is stands at the moment.

For FM, in automatic mode, it always seems to set a value of 1, so commenting out the call should result in no change of behaviour.

You could add an option for LW/MW to select between a loop antenna (0 = automatic mode) and whip antenna (1 = non-automatic minimum capacitance), but I think this will add further confusion.

Please checkout the behaviour yourself and hopefully you will come to the same conclusions.

73, Dave

max-arnold, 2025-03-21:
I think I found where this memory effect (the dependence on a previous band) comes from: https://github.com/pu2clr/SI4735/blob/master/src/SI4735.cpp#L765-L779

The `setAM` call does a powerDown/powerUp cycle, but only if the mode has changed. This explains why the VHF->MW1 switch or 80m->MW3 switch resets the capacitor (the modes are different) and MW1->MW2 doesn't (the mode is the same).

I temporarily patched the SI4735 library to do the power cycle unconditionally, and the memory effect has gone. The capacitor value always reads as 1 even in automatic mode (I did apply the getFrequency fix), and the RSSI doesn't drop.

So I think it is safe to set it to 1 for the MW band

max-arnold, 2025-03-21:
Fixed in v1.07

G8PTN, 2025-03-21:
Hi Max,
I think all the modes may cause a power cycle, and this may be necessary for the correct operation of the device. You would need to carefully read through the manual and the notes in the PU2CLR code.

I would not expect the capacitor value always to be 1 in automatic mode for AM.

As stated previously if the call rx.setTuneFrequencyAntennaCapacitor(0); is made after the mode change, it would work after a power cycle. The auto tuning is only valid for a loop antenna configuration.

With the loop antenna, the values I measured were quite close to theory, based on the Inductance of the supplied loop antenna. With a whip antenna, a resonant point will not be reached and "C" will max out at 6143, which is what causes the attenuation when using the whip.

73, Dave

[image]

[image]

max-arnold, 2025-03-21:
I tried moving the `rx.setTuneFrequencyAntennaCapacitor()` call after the mode change like this:

```diff
diff --git a/ats-mini/ats-mini.ino b/ats-mini/ats-mini.ino
index 76d721f..8cb1f15 100644
--- a/ats-mini/ats-mini.ino
+++ b/ats-mini/ats-mini.ino
@@ -1110,7 +1110,6 @@ void useBand() {
   currentMode = bandMODE[bandIdx];                  // G8PTN: Added to support mode per band
   if (band[bandIdx].bandType == FM_BAND_TYPE) {
     currentMode = FM;
-    // rx.setTuneFrequencyAntennaCapacitor(0);
     rx.setFM(band[bandIdx].minimumFreq, band[bandIdx].maximumFreq, band[bandIdx].currentFreq, tabFmStep[band[bandIdx].currentStepIdx]);
     rx.setSeekFmLimits(band[bandIdx].minimumFreq, band[bandIdx].maximumFreq);
     bfoOn = ssbLoaded = false;
@@ -1121,9 +1120,8 @@ void useBand() {
     rx.setRdsConfig(1, 2, 2, 2, 2);
     rx.setGpioCtl(1,0,0);   // G8PTN: Enable GPIO1 as output
     rx.setGpio(0,0,0);      // G8PTN: Set GPIO1 = 0
+    rx.setTuneFrequencyAntennaCapacitor(0);
   } else {
-    // set the tuning capacitor for SW or MW/LW
-    // rx.setTuneFrequencyAntennaCapacitor((band[bandIdx].bandType == MW_BAND_TYPE || band[bandIdx].bandType == LW_BAND_TYPE) ? 0 : 1);
     if (ssbLoaded) {
       // Configure SI4732 for SSB
       rx.setSSB(
@@ -1156,6 +1154,8 @@ void useBand() {
     rx.setGpio(1,0,0);      // G8PTN: Set GPIO1 = 1
     rx.setSeekAmLimits(band[bandIdx].minimumFreq, band[bandIdx].maximumFreq); // Consider the range all defined current band
     rx.setSeekAmSpacing(5); // Max 10kHz for spacing
+    // set the tuning capacitor for SW or MW/LW
+    rx.setTuneFrequencyAntennaCapacitor((band[bandIdx].bandType == MW_BAND_TYPE || band[bandIdx].bandType == LW_BAND_TYPE) ? 0 : 1);
   }
 
   // G8PTN: Added
@@ -3209,7 +3209,7 @@ void loop() {
     Serial.print(currentBFO);                   // Frequency (Hz x 1000)
     Serial.print(",");
 
-    Serial.print(bandIdx);                      // Band
+    Serial.print(band[bandIdx].bandName);       // Band
     Serial.print(",");
     Serial.print(currentMode);                  // Mode
     Serial.print(",");
```

And here is the result with a whip antenna:

```
107,8650,0,VHF,0,1,0,4,4,0,26,2,1,2837,25
107,8650,0,VHF,0,1,0,4,4,0,26,2,1,2837,26
107,8650,0,VHF,0,1,0,4,4,0,26,2,1,2837,27
107,800,0,MW1,3,3,0,4,4,0,26,1,6143,2842,28
107,800,0,MW1,3,3,0,4,4,0,26,11,6143,2842,29
107,800,0,MW1,3,3,0,4,4,0,26,10,6143,2842,30
107,800,0,MW1,3,3,0,4,4,0,26,11,6143,2842,31
107,765,0,MW2,3,2,0,4,4,0,26,8,6143,2843,32
107,765,0,MW2,3,2,0,4,4,0,26,12,6143,2843,33
107,765,0,MW2,3,2,0,4,4,0,26,10,6143,2843,34
107,765,0,MW2,3,2,0,4,4,0,26,9,6143,2843,35
107,2500,0,MW3,3,1,0,4,4,0,26,21,6143,2844,36
107,2500,0,MW3,3,1,0,4,4,0,26,22,6143,2844,37
107,2500,0,MW3,3,1,0,4,4,0,26,21,6143,2844,38
107,3700,0,80M,1,0,0,4,4,0,26,21,6143,2844,39
107,3700,0,80M,1,0,0,4,4,0,26,23,6143,2846,40
107,3700,0,80M,1,0,0,4,4,0,26,25,6143,2846,41
107,3700,0,80M,1,0,0,4,4,0,26,23,6143,2845,42
107,2500,0,MW3,3,1,0,4,4,0,26,21,6143,2846,43
107,2500,0,MW3,3,1,0,4,4,0,26,23,6143,2846,44
107,765,0,MW2,3,2,0,4,4,0,26,4,6143,2846,45
107,765,0,MW2,3,2,0,4,4,0,26,8,6143,2847,46
107,765,0,MW2,3,2,0,4,4,0,26,9,6143,2847,47
107,800,0,MW1,3,3,0,4,4,0,26,11,6143,2846,48
107,800,0,MW1,3,3,0,4,4,0,26,12,6143,2846,49
107,8650,0,VHF,0,1,0,4,4,0,26,3,1,2844,50
107,8650,0,VHF,0,1,0,4,4,0,26,3,1,2844,51
107,8650,0,VHF,0,1,0,4,4,0,26,4,1,2844,52
107,8650,0,VHF,0,1,0,4,4,0,26,4,1,2844,53
```

For comparison, this is how it looks right now with no calls to `rx.setTuneFrequencyAntennaCapacitor` (RSSI is higher):

```
107,8650,0,VHF,0,1,0,4,4,0,26,2,1,2862,12
107,8650,0,VHF,0,1,0,4,4,0,26,2,1,2864,13
107,800,0,MW1,3,3,0,4,4,0,26,41,1,2869,14
107,800,0,MW1,3,3,0,4,4,0,26,42,1,2869,15
107,800,0,MW1,3,3,0,4,4,0,26,40,1,2869,16
107,800,0,MW1,3,3,0,4,4,0,26,40,1,2869,17
107,765,0,MW2,3,2,0,4,4,0,26,40,1,2870,18
107,765,0,MW2,3,2,0,4,4,0,26,42,1,2870,19
107,765,0,MW2,3,2,0,4,4,0,26,40,1,2869,20
107,2500,0,MW3,3,1,0,4,4,0,26,51,1,2870,21
107,2500,0,MW3,3,1,0,4,4,0,26,51,1,2870,22
107,2500,0,MW3,3,1,0,4,4,0,26,51,1,2870,23
107,3700,0,80M,1,0,0,4,4,0,26,44,1,2868,24
107,3700,0,80M,1,0,0,4,4,0,26,50,1,2868,25
107,3700,0,80M,1,0,0,4,4,0,26,53,1,2868,26
107,2500,0,MW3,3,1,0,4,4,0,26,51,1,2872,27
107,2500,0,MW3,3,1,0,4,4,0,26,51,1,2872,28
107,765,0,MW2,3,2,0,4,4,0,26,42,1,2871,29
107,765,0,MW2,3,2,0,4,4,0,26,40,1,2871,30
107,800,0,MW1,3,3,0,4,4,0,26,42,1,2870,31
107,800,0,MW1,3,3,0,4,4,0,26,41,1,2870,32
107,8650,0,VHF,0,1,0,4,4,0,26,2,1,2869,33
107,8650,0,VHF,0,1,0,4,4,0,26,3,1,2869,34
107,8650,0,VHF,0,1,0,4,4,0,26,2,1,2869,35
```

max-arnold, 2025-03-21:
> I would not expect the capacitor value always to be 1 in automatic mode for AM.

I disabled the automatic mode as you suggested earlier:

> If you comment out the calls rx.setTuneFrequencyAntennaCapacitor(0); the capacitor value should always be equal to 1 (non-automatic, minimum capacitance). Whilst this is not the ideal point for a [...]

max-arnold, 2025-03-21:
>  I think all the modes may cause a power cycle, and this may be necessary for the correct operation of the device. You would need to carefully read through the manual and the notes in the PU2CLR code.

Unfortunately, the power cycle handling is different for `setAM`, `setFM`, and `setSSB`. But it is only the `setAM` call where it is conditional based on the previous mode. This is what causes the "memory effect" when RSSI depends on the bands you were visiting before. I think it is a bug.

G8PTN, 2025-03-21:
Hi Max,
For reference, this is the change that I made a couple of weeks ago when I was looking at the issue.
I also removed the redundant white spaces.
By default, the rx.setTuneFrequencyAntennaCapacitor(0); is commented out, so in all cases the return value should be 1.

[useband_patch.txt](https://github.com/user-attachments/files/19388263/useband_patch.txt)

73, Dave

max-arnold, 2025-03-21:
I'm a bit confused :) Does it mean that you are happy with the current implementation (i.e. both calls to `setTuneFrequencyAntennaCapacitor` being commented out)?

G8PTN, 2025-03-21:
Hi Max,
Yes, I think both calls should be commented out to give acceptable performance for all antenna types.
73, Dave

closed 2025-03-22 by G8PTN

G8PTN, 2025-05-12:
Hi Max,
With the tuning capacitor enabled, as discussed previously, I would expect the signal levels to reduce when using a whip antenna on LW/MW bands.

Also, I am not sure what the behaviour will be with the hardware with the FET on the front end.

[image]

73, Dave

max-arnold, 2025-05-12:
Yes, I tested the capacitor with a whip antenna and indeed the signal levels are reduced. With a loop antenna the capacitor actually works! And I fixed the annoying "memory effect" when it was dependent on a previous modulation type.

I marked this feature as experimental and if I get some negative feedback with the whip antennas, I'll have to add a setting for that (possibly a per-band one)

G8PTN, 2025-05-13:
Hi Max,
All understood.

With hardware V3, that has the JFET on the AM input, the auto capacitor will not work even for a loop antenna, because the antenna is effectively isolated from the SI4732 AMI input pin. I suspect the capacitor will get set to the maximum value, but this may not degrade the signal too much because the JFET can drive a low impedance.

I don't think the auto capacitor behaviour was considered when the JFET buffer was added. The JFET provides a broadband impedance transformation, but it will not have the selectivity of the original SI4732 tuned circuit arrangement.

73, Dave

max-arnold, 2026-11-09:
I'm pretty sure I will regret it again, but I added an optional Wi-Fi only setting to re-enable the capacitor on receivers without the high Z circuit. See 088970039d64