esp32-si4732/ats-mini #15: Maybe get rid of 1M step on SSB

max-arnold, 2025-03-21:
* Separate step tables for SSB/AM?
* Related discussion https://github.com/G8PTN/ATS_MINI/issues/6#issuecomment-2694879710
* Maybe just store the step per mode (not per mode per band)?

G8PTN, 2025-03-21:
Hi Max,
A 1MHz step for SSB is not a valid option anyway.
I corrected the comments in the code.
[image]

I agree separate configuration could be added for AM/SSB step (per band), but the same is also true for bandwidth.
[image]

The whole concept of the band table is a little messy and probably a rethink is required to have a better solution. However, since many of the ATS based receivers operate in this way, which is not ideal, changing may become confusing to users.

Although the UI is now much better, I think the single control is not ideal, on one of my receivers, the encoder is now broken, and I will need swap it out. Perhaps the encoder should be a non-detent type, although these are more prone to accidental adjustments.

For the ATS20+ receiver, I created a simple remote Bluetooth App, which allows all controls to be done remotely. Most of the other forks of the ATS-Mini are adding WIFI control and monitoring, the only disadvantage is the current consumption and RFI, although the current consumption is not a real problem for home use, where the receiver can be permanently powered.

Keep up the great work and I will keep an eye on future updates.

73, Dave

G8PTN, 2025-03-22:
Hi Max,

All understood.

WIFI is great for Webserver based solutions and supports any device type.

BLE should be much less power, but the ESP32-S3 Bluetooth (BLE) radio is not so power efficient. There may be power saving options that can be enabled for BLE.

For BT or WIFI, the ATS-Mini should have a menu option to turn off the radio to conserve power when remote control is not being used.

I used the MIT App Inventor tool for the ATS20+ controller and a UNI-T multi-meter that I have. The MIT tool is easy to use for people like me that have never created an App.

For the ATS20+, I fitted a cheap JDY-31 Bluetooth module, which has an average operating current of 7mA, which is significantly lower than the ESP32-S3.

73, Dave

[image]

luarvique, 2025-04-27:
This issue has been addressed with the latest round of refactoring. Please, test whatever is currently in the main branch and let me know if this issue can be closed.

closed 2025-04-28 by max-arnold