esp32-si4732/ats-mini #17: Per Band Tables

G8PTN, 2025-03-21:
Hi Max,
I see there is a pull request, where someone has updated the band table to add another VHF band.

The other per band tables (bandCAL and bandMODE) need to be updated according, otherwise there will be mismatches and array elements will be indexed that do not exist.

The additional tables could be added to the main table, provided the code is updated according, but they were initially kept separate to maintain memory map compatibility with the original (stock) version of the firmware.

In the case of FM_BAND_TYPE, it will actually force the mode to FM, but the other modes will be incorrect if the tables are not updated accordingly.

Sorry for any confusion.
73, Dave

[image]

max-arnold, 2025-03-22:
No worries! I'm just encouraging other people to experiment with their own forks, and this PR was likely produced by GitHub when the author used its web UI to make changes

DXKolpino, 2025-03-22:
Здравствуйте, Дэйв

Извините, это я постигаю сложную науку создания прошивок для Малышарика.
Спасибо за науку.

С уважением, Никита

G8PTN, 2025-03-22:
Hi Nikita,
No problem. The best way to understand and learn is by experimenting.
Kind Regards,
  Dave

closed 2025-03-22 by G8PTN