esp32-si4732/ats-mini #23: AGC does not work properly in SSB until you turn it on in AM mode

K2COW, 2025-03-25:
There are several reports from users where AGC does not work properly ( does not take effect) in SSB until you turn it on in AM mode and then come back to SSB

max-arnold, 2025-03-26:
I need a step-by-step example of how to reproduce the problem (maybe a short video?)

K2COW, 2025-03-26:
I'm still waiting on my device to arrive, others have stated that when they boot the device and go into LSB or USB and try to enable AGC there is no effect. When they switch to AM and toggle on the AGC it starts working and when they switch back to USB or LSB then AGC works.

max-arnold, 2025-03-26:
Does it happen with any firmware, or with a specific version?

> when they boot the device and go into LSB or USB and try to enable AGC

Strange... AGC seems to be enabled by default

G8PTN, 2025-03-26:
Hi,
My understanding is that the SSB patch for the SI4732/5 only appears to support two options and so the settings on the ATS-Mini have been limited accordingly.

The AGC setting ranges for the different modes are shown in the table below. The settings are stored on a per mode basis.

[image]

ATTN = 0 is “ACG OFF” with no attenuation.
The SI4732/5 SSB patch effectively only supports “AGC ON” and “AGC OFF”.

In SSB mode, with the AGC enabled, there will be a low-level tone in the background, even with the volume set to low. This is a known issue with the SSB patch and there is no workaround other than disabling the AGC.

There are various threads on other forums that discuss these issues.
https://groups.io/g/si47xx/topic/si4732_4735_agc_working/95455063
https://github.com/pu2clr/SI4735/issues/15

An option is added on ATS-MINI to adjust the AVC gain values for AM and SSB, but this operates at the audio stage rather than the RF stage for the AGC, but may be useful in some cases.

[image]

73, Dave

K2COW, 2025-03-26:
Thank you so much for your very detailed reply Dave

max-arnold, 2025-04-15:
@K2COW Could you please try the following build to see whether it fixes the AGC issue? https://github.com/esp32-si4732/ats-mini/actions/runs/14465651796

max-arnold, 2025-04-19:
The AGC change went into the 2.10 release

closed 2025-04-28 by max-arnold