esp32-si4732/ats-mini #24: General observations.

G8PTN, 2025-03-26:
Hi Max,
A few general observations.

- Push and rotate
The short, medium and long push works fine, but the push and rotate is not particularly easy to use in my opinion.

- Power up rebooting
I have noticed with my ATS_MINI firmware after compiling (only a few occurrences), on power up the receiver seems to reboot several times. This generally happens when there is no USB connected. I am not sure if this is a bug in the code, hardware, serial port or compiler related. I have also seen similar issues with the debug print statements enabled.

    There are various threads that report possible issues with the CDC serial port on the ESP32-S3, but I have not seen anything conclusive.

    On my version of the code, I have added the remote log enable/disable that you have and defaulted it to off as a counter measure. I have not seen any issues so far, but that may be due to the recompile!

- Hardware
    On the two receivers that I have, one now has a defective rotary encoder, and both exhibit an issue with the ON/OFF switch. The contact resistance of the switch seems to be very inconsistent and gets worse over time. I noticed that the battery voltage intermittently goes up and down and this is due to the ON/OFF switch. I have seen the same problem on most devices that use this type of switch.

    I will probably modify the units to use a MOSFET power switch, so that it does not depend on the quality of the ON/OFF switch itself.

    I expect that over time, both the rotary encoder and ON/OFF switch will be a problem on these receivers.

73, Dave

max-arnold, 2025-03-26:
> the push and rotate is not particularly easy to use in my opinion.

I kind of agree. Push + rotate are quite usable for seeking up/down on AM/FM (you need to turn the encoder just a bit to trigger the process). On SSB this is more like a kludge. People were asking for faster tuning (e.g. an acceleration factor for the encoder), and push + rotate was the best idea I came up with so far. I doubt that the acceleration factor will work on this cheap encoder.

Another option I was thinking about is using push+rotate to select the tuning step (the step menu will be dismissed immediately after you release the encoder). But this involves more clicks, especially if you want to get back to a smaller step.

How about the following approach (SSB only): push + rotate will switch to a predefined faster step (as it does now with the lookup table) and as long as you rotate the knob you can stop pushing it. After a certain period of no activity the step will switch back to the initial value.

By the way, I replaced the stock knob with this one (it is much bigger and fits really well) https://www.aliexpress.com/item/1005007003511755.html  I can push it with one hand and rotate with the other :)

> Power up rebooting

Happened once or twice for me. I suspect that we need to add some delays to the early initialization code. Or this could be a power consumption spike.

Another idea is to turn the serial port into a menu setting that is disabled by default.

> On the two receivers that I have, one now has a defective rotary encoder

The same for me. And on the second receiver the USB port is not soldered well and sometimes drops the connection if I touch the cable.  But this is expected giving the toy nature and price of this radio

max-arnold, 2025-03-26:
> I will probably modify the units to use a MOSFET power switch, so that it does not depend on the quality of the ON/OFF switch itself.

If you do that, how about expanding the [Hardware notes](https://esp32-si4732.github.io/ats-mini/hardware.html) page? Or adding a separate page dedicated to simple hardware mods.

G8PTN, 2025-03-26:
Hi Max
> Push and rotate

Yes, I agree the action to trigger the search is not too bad.
I think the option of switching to the predefined faster step might be a good option, but could still be confusing.
It is one of those things to try out and get opinions from users.

> Power up rebooting

It may be a power spike, although I have had a quick look on the oscilloscope and the VCC to the ESP32-S3 look reasonable, and sometimes it cycles a few times. I tried adding some delays after setting the serial port, but it did not make any difference.

I thought it might be the GPIO0 (Boot) pin since it relies on the internal pull-up, but the signal looks good, but I have not checked the timing between GPIO0 and EN. If it sampled GPIO0 low it should just go into bootloader mode.

There is quite a high in-rush current at power up, but I don't see why that should be compile dependant, unless there is some subtle timing differences.

Like most things, because it is not repeatable and depends on the compile, it makes it a little difficult to diagnose.
I will let you know if I find anything of note.

> Hardware mods

Yes, perhaps add a page or section headings for this.

73, Dave

MattMan76, 2025-03-26:
My thought on this is to have the seekUp seekDown back as menu options and then as long as the menu returns to the last choice you can just double press the button to seek again.

G8PTN, 2025-03-26:
Hi Matt,
I think there are pros and cons with all the methods.

I went for the keep it KISS (keep it simple, stupid approach), with single press and always reset the menu to a known position (Volume), working on the principle that all the commonly used options were then +/- 2 rotates away.

Returning to the previous menu is perhaps better, but it just means that then when you want to change something else it is more steps. The medium push sort of helps because it provides a shortcut to the volume adjustment and also resets the menu to the volume selection.

I think the push and single rotate action is not so bad, it is the push and continuous rotate that seems fiddly to me.

I did wonder if making the long press for backlight OFF could be changed to a menu option and then use long press as a shortcut for step selection.

Unfortunately, with the single control available, everyone will probably have a different preference for their use case.

I did think of making a wired USB based remote panel for home use, with its own tuning control and separate buttons for all the common actions, but I will probably go down the WIFI or BT App approach, since for home use the increased power for the ESP32-S3 radio is not so critical, it can be permanently powered, and then use the App for everything. The only thing really missing is BT audio which is not natively supported by the ESP32-S3, so an additional chip (module) or external adapter would be needed.

The main advantage of the ATS-Mini is its small size, which is very convenient just to put in your pocket for portable use. A simple headphone amplifier should probably have been incorporated on the PCB to make the headphone jack more usable.

73, Dave