esp32-si4732/ats-mini #40: Serial command `t` or `T` will increase volume to maximum

andimik, 2025-04-05:
I just wanted to get some data out via serial port and realized, that the command `t` or `T` will continousely increase the volume till 63. 

I don't know if this is intended.

```
$ cat /dev/ttyACM0 
VHF,0,0,0,4,4,22,37,3,1,2587,8

109,10600,0,VHF,0,0,0,4,4,22,37,2,1,2587,9

109,10600,0,VHF,0,0,0,4,4,22,37,1,1,2588,10

109,10600,0,VHF,0,0,0,4,4,22,37,2,1,2588,11

109,10600,0,VHF,0,0,0,4,4,22,37,2,1,2591,12

109,10600,0,VHF,0,0,0,4,4,22,37,3,1,2591,13

109,10600,0,VHF,0,0,0,4,4,22,38,3,1,2591,14

109,10600,0,VHF,0,0,0,4,4,22,39,5,1,2590,15

109,10600,0,VHF,0,0,0,4,4,22,40,3,1,2590,16

109,10600,0,VHF,0,0,0,4,4,22,41,2,1,2593,17

109,10600,0,VHF,0,0,0,4,4,22,42,1,1,2591,18

109,10600,0,VHF,0,0,0,4,4,22,43,3,1,2590,19

109,10600,0,VHF,0,0,0,4,4,22,44,10,1,2593,20

109,10600,0,VHF,0,0,0,4,4,22,45,9,1,2591,21

109,10600,0,VHF,0,0,0,4,4,22,46,2,1,2592,22

109,10600,0,VHF,0,0,0,4,4,22,47,2,1,2593,23

109,10600,0,VHF,0,0,0,4,4,22,48,2,1,2591,24

109,10610,0,VHF,0,0,0,4,4,22,49,0,1,2588,25

109,10625,0,VHF,0,0,0,4,4,22,50,0,1,2588,26

109,10635,0,VHF,0,0,0,4,4,22,51,0,1,2589,27

109,10640,0,VHF,0,0,0,4,4,22,52,0,1,2590,28

109,10640,0,VHF,0,0,0,4,4,22,53,0,1,2586,29

109,10640,0,VHF,0,0,0,4,4,22,54,0,1,2592,30

109,10640,0,VHF,0,0,0,4,4,22,55,0,1,2587,31

109,10640,0,VHF,0,0,0,4,4,22,56,0,1,2580,32

109,10650,0,VHF,0,0,0,4,4,22,57,0,1,2575,33

109,10650,0,VHF,0,0,0,4,4,22,58,0,1,2579,34

109,10650,0,VHF,0,0,0,4,4,22,59,0,1,2568,35

109,10665,0,VHF,0,0,0,4,4,22,60,0,1,2578,36

109,10680,0,VHF,0,0,0,4,4,22,61,0,1,2577,37

109,10690,0,VHF,0,0,0,4,4,22,62,0,1,2567,38

109,10690,0,VHF,0,0,0,4,4,22,63,0,1,2584,39

109,10690,0,VHF,0,0,0,4,4,22,63,0,1,2567,40

109,10690,0,VHF,0,0,0,4,4,22,63,0,1,2554,41
```

max-arnold, 2025-04-08:
No, it is not intended and I can't reproduce it.  I use the following command:

```
picocom /dev/tty.usbmodem14401 -b115200
```

andimik, 2025-04-10:
I see, with your code the volume increase does not happen.

I've described my commands here: https://www.rundfunkforum.de/viewtopic.php?p=1802339#p1802339

# For the screenshot:

First set the port writeable

`sudo chmod o+rw /dev/ttyACM0`

Then prepare for step 3 

`cat /dev/ttyACM0 > /tmp/dump.raw`

Then send a capital C

`echo -e "C" > /dev/ttyACM0 `

At the end, change it to bitmap

`xxd -r -p < /tmp/dump.raw > /tmp/dump.bmp`

# For the `t` or `T` problem:

`echo -e "T" > /dev/ttyACM0 `

max-arnold, 2025-04-17:
I reproduced the problem, but I have no idea what causes it (maybe some `stty` options are needed?). But you can use the following oneliner that works:

```shell
echo -n C | socat stdio /dev/cu.usbmodem14401,echo=0,raw | xxd -r -p > /tmp/screenshot.bmp
```

I also fixed the screenshot palette.

A test build is available here https://github.com/esp32-si4732/ats-mini/actions/runs/14512258762 (not released yet)

closed 2025-04-18 by max-arnold

andimik, 2025-04-18:
yes, this works now, thanks!

[screenshot.bmp.zip](https://github.com/user-attachments/files/19811992/screenshot.bmp.zip)