esp32-si4732/ats-mini #41: V1.09 - Display rotated and inverted

jimjackii, 2025-04-08:
I received a new radio and immediately installed version 1.09 - unfortunately the display is now rotated and inverted. What could be the cause of this? Reinstalling the firmware didn't help either.

Unfortunately, I had forgotten to back up the old firmware.

Regards, Steffen

G8PTN, 2025-04-08:
Hi,
If you are able to recompile the code, it should be possible to make the changes detailed in the following thread, which is a patch to the ATS_MINI firmware.

https://github.com/G8PTN/ATS_MINI/issues/24#issuecomment-2785986783

I believe that Max may be looking at an automatic display detection update for the ats-mini firmware.

Kinds Regards,
  Dave

jimjackii, 2025-04-08:
Hi Dave,

thanks for the tip. Unfortunately I had overlooked that.

Regards, Steffen

max-arnold, 2025-04-08:
@jimjackii @BrightCGN @bea5t I made an experimental fix for the mirrored display issue. You can try the following build (see the ats-mini-v1.09-3-xxx artifact): https://github.com/esp32-si4732/ats-mini/actions/runs/14332986320

If it works for you, please let me know and also post a photo of the Settings->About screen.

bea5t, 2025-04-08:
THX, it works!

[image]

jimjackii, 2025-04-08:
Thank you, Max.

I'll test it tonight and get back to you with the results.

Regards, Steffen

BrightCGN, 2025-04-08:
Hello Max,

it workls for me :-)

G8PTN, 2025-04-08:
Hi Max,
The readback value is similar to that in the datasheet for the GC9307 driver chip.

[image]

Is this just the ID of the driver chip?

I think the actual behaviour will depend on the driver and display combination, so I am not sure basing the selection on the RDDID is always going to work.

73, Dave

max-arnold, 2025-04-08:
> I am not sure basing the selection on the RDDID is always going to work.

I 100% agree, but it might be a good short-term solution. I can introduce a setting (via the menu or serial port) later.

max-arnold, 2025-04-08:
> it workls for me :-)

@BrightCGN Please post a photo of the Settings > About screen. It is needed to collect more data for a fix that will go into the next stable firmware version.

jimjackii, 2025-04-08:
The test version is also fine for me.

Regards, Steffen

[image]

gutao3800cc, 2025-04-10:
我说中文吧，我对此有一点经验，GC9307除了屏幕翻转基本可以和ST7789一样，我们只需要修改ST7789的驱动文件。
2024/12/19

更正原理图，有的esp32 GPIO0有抖动现象。修改电路板，手动拉高io0。

修改代码，音量为0时，RDA5807与NS4160同时静音，可能会省一点电？

适配GC9307驱动的屏幕，此屏便宜一些，但需要轻微打磨外壳屏幕框。

需要修改tft_espi驱动文件 路径：C:\Users\你电脑用户名\Documents\Arduino\libraries\TFT_eSPI\TFT_Drivers\ST7789_Rotation.h

第4行  rotation = m % 4; 修改为rotation = m % 5; 在最后一个break；后加入

case 4: // Inverted landscape
#ifdef CGRAM_OFFSET
      if (_init_width == 135)
      {
        colstart = 40;
        rowstart = 52;
      }
      else if(_init_height == 280)
      {
        colstart = 20;
        rowstart = 0;
      }
      else if(_init_width == 172)
      {
        colstart = 0;
        rowstart = 34;
      }
      else if(_init_width == 170)
      {
        colstart = 0;
        rowstart = 35;
      }
      else
      {
        colstart = 80;
        rowstart = 0;
      }
#endif
      writedata( TFT_MAD_MV | TFT_MAD_MX | TFT_MAD_MY | TFT_MAD_COLOR_ORDER);

      _width  = _init_height;
      _height = _init_width;
      break;
然后在主程序修改为tft.setRotation(4);就可以正常显示了，但是颜色是反的，我没有精力去修改。

max-arnold, 2025-04-10:
@gutao3800cc Thanks!

I fixed it slightly differently without the need to modify the library, see the following change: https://github.com/esp32-si4732/ats-mini/compare/main...display-fix

max-arnold, 2025-04-15:
So I added an automatic detection of the display type. Could you try the following build? https://github.com/esp32-si4732/ats-mini/actions/runs/14462363024

BrightCGN, 2025-04-15:
> So I added an automatic detection of the display type. Could you try the following build? https://github.com/esp32-si4732/ats-mini/actions/runs/14462363024

Works for me very fine 👍

jimjackii, 2025-04-15:
Thank you, Max.

I'll test it tonight.

jimjackii, 2025-04-15:
I am running the test version V2.0, thank you.

I haven't tested the normal version of Github yet.

Regards, Steffen

heibertelf, 2025-04-16:
> So I added an automatic detection of the display type. Could you try the following build? https://github.com/esp32-si4732/ats-mini/actions/runs/14462363024

This fixed the flipped display, but it seems like the colors are wrong - for example, the orange theme looks like a blue. 
But the colors are a smaller problem, at least the screen is readable.
Thanks for your work!

max-arnold, 2025-04-17:
@heibertelf Could you try the following build (see the ats-mini-v1.09-3-xxx artifact): https://github.com/esp32-si4732/ats-mini/actions/runs/14332986320

If it works for you, please let me know and also post a photo of the Settings->About screen.

heibertelf, 2025-04-17:
> [@heibertelf](https://github.com/heibertelf) Could you try the following build (see the ats-mini-v1.09-3-xxx artifact): https://github.com/esp32-si4732/ats-mini/actions/runs/14332986320
> 
> If it works for you, please let me know and also post a photo of the Settings->About screen.

No difference with previous build. Screen orientation is ok, but orange theme still blue.

[image]

[image]

max-arnold, 2025-04-18:
@heibertelf Could you try the following build? https://github.com/esp32-si4732/ats-mini/actions/runs/14535957017

jimjackii, 2025-04-18:
Hi Max,

here my colors for ths new testversion.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

max-arnold, 2025-04-19:
Ok, here is a new debug build to collect more data https://github.com/esp32-si4732/ats-mini/actions/runs/14546732857. The autodetection logic is disabled, so the proper image will be only shown on the "classic" 7789 displays.

You need to go to the `Settings -> About` screen and post a photo of it here. Mine looks like this

[image]
DID: 048181B3, DST: 09D0532400

jimjackii, 2025-04-19:
Hello Max,

[image]

this is a test with the other display version (ID=93), but maybe this will help you.

Regards, Steffen

BrightCGN, 2025-04-19:
Here  a scrennshot of mine wth the newest build verison.

[image]

max-arnold, 2025-04-19:
@BrightCGN Richard, your screen seems ok - the display model is 9307, but the image looks normal. It is likely that you are using the released version. Can you try the following debug build instead? https://github.com/esp32-si4732/ats-mini/actions/runs/14546732857

BrightCGN, 2025-04-19:
> [@BrightCGN](https://github.com/BrightCGN) Richard, your screen seems ok - the display model is 9307, but the image looks normal. It is likely that you are using the released version. Can you try the following debug build instead? https://github.com/esp32-si4732/ats-mini/actions/runs/14546732857

I instaled it and this is the result

[image]

max-arnold, 2025-04-19:
@BrightCGN @jimjackii Please one more test with the About screen https://github.com/esp32-si4732/ats-mini/actions/runs/14549704060

max-arnold, 2025-04-19:
And here is how the themes should look like:

[image]

BrightCGN, 2025-04-19:
> [@BrightCGN](https://github.com/BrightCGN) [@jimjackii](https://github.com/jimjackii) Please one more test with the About screen https://github.com/esp32-si4732/ats-mini/actions/runs/14549704060

Tested it with the new version.
Now rotation works.
Also the orange theme is working for me.

Thanks for your great work.

[image]

[image]

jimjackii, 2025-04-19:
Hi Max,
this time it also looks very good with the display ID=93.
Plus a few more themes.
Regards, steffen

[image]

[image]

[image]

[image]

max-arnold, 2025-04-20:
Great! So the conclusion is that there are two display types discovered so far:

* ST7789 - the original one
* GC9307 - vertically mirrored, colors are inverted, RGB order swapped

The fix that autodetects the display model will be released in 2.11

G8PTN, 2025-04-20:
Hi Max,
I think the ST7789 is effectively BGR as well.

I have been using the original TFT_eSPI library from the ralphxavier information as previously covered in the discussion thread.
https://github.com/esp32-si4732/ats-mini/discussions/31#discussion-8148752

I am not sure what settings you have in the Setup206_LilyGo_T_Display_S3.h, but for whatever reason the behaviour seems to depend on the INIT_SEQUUENCE_3.

I tried reading the Display ID (DID) and Display Status (DST) registers with the older TFT_eSPI library that is covered in the discussion thread and noticed that it read back all zeros, so I switched to a more up to date library, which allows the registers to be read.

I did a few tests cases using the ats-mini V2.10 code in the Arduino IDE setup that I have.
 
What seems strange is that test cases 1 and 3 produce the correct display colours (see below), but the DST readback value is different. All tests are with the same unit, normal ST7789 type.

I still don’t understand why the INIT_SEQUUENCE_3 changes the colour order behaviour.

Provided your automatic workaround behaves as expected, I wouldn't worry too much.

73, Dave

<Test Case 1>
Setup206_LilyGo_T_Display_S3.h
#define INIT_SEQUENCE_3
#define TFT_RGB_ORDER TFT_RGB

ats-mini.ino (V2.10)
tft.setRotation(3);
tft.invertDisplay(1);
[image]
Display shows correct colours.
DST = 09D0532400 (2nd parameter D2 = 0 = RGB), but inferred behaviour is as BGR.

<Test Case 2>
Setup206_LilyGo_T_Display_S3.h
//#define INIT_SEQUENCE_3
#define TFT_RGB_ORDER TFT_RGB

ats-mini.ino (V2.10)
tft.setRotation(3);
tft.invertDisplay(1);
[image]
Display now shows incorrect colours, but no change in DST readback values.
DST = 09D0532400 (2nd parameter D2 = 0 = RGB), not changed.

<Test Case 3>
Setup206_LilyGo_T_Display_S3.h
//#define INIT_SEQUENCE_3
#define TFT_RGB_ORDER TFT_RGB

ats-mini.ino (V2.10)
tft.setRotation(3);
tft.invertDisplay(1);

tft.writecommand(TFT_MADCTL);
tft.writedata(TFT_MAD_MV | TFT_MAD_MY | TFT_MAD_BGR);  
[image]
Display shows correct colours, DST readback value changes
DST = 09D4532400 (2nd parameter D2 = 1 = BGR)

max-arnold, 2025-04-20:
@G8PTN Thanks for the additional tests! One thing I noticed is that the DID for your display is `04858552`. For mine it is different - `048181B3`.

> I am not sure what settings you have in the Setup206_LilyGo_T_Display_S3.h, but for whatever reason the behaviour seems to depend on the INIT_SEQUUENCE_3.

My build process uses `arduino-cli` and is fully repeatable via the `ats-mini/sketch.yaml` config file that pins the exact library versions with no customizations whatsoever. The TFT_eSPI library is configured in the project itself via the `ats-mini/tft_setup.h` file.  This means that the `Setup206_LilyGo_T_Display_S3.h` is not modified in any way (it is part of the library)

max-arnold, 2025-04-20:
@Rfromearth Before I include your fix, could you please do a test with this specific fw build: https://github.com/esp32-si4732/ats-mini/actions/runs/14554673801

Flash it and then post a photo of the Settings->About screen here.

G8PTN, 2025-04-20:
Hi Max,
Yes, all understood. You are using the current (latest) TFT_eSPI library which has the following settings.

#define INIT_SEQUENCE_3 // Using this initialisation sequence improves the display image
#define CGRAM_OFFSET
#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//#define TFT_RGB_ORDER TFT_BGR // Colour order Blue-Green-Red
#define TFT_INVERSION_ON
// #define TFT_INVERSION_OFF

The ST7789 has a register that does not exist for the GC9307 called LCMCTRL. 

This can invert the MADCTL bits. One of the bits that gets set to invert is the RGB color mode when #define INIT_SEQUENCE_3 is enabled. This is set in  ST7789_Init.h

When the GC9307 driver chip is used, this is why you have to make the setting TFT_MAD_BGR, because the library will be setting TFT_MAD_COLOR_ORDER = TFT_MAD_RGB which is correct for the ST7789 driver chip, but will be incorrect for the GC9307.

In conclusion, the experimental approach is probably appropriate to determine the correct settings for the ST7789 and GC9307 display types, since the behaviour depends on which TFT_eSPI library that is being used.

73, Dave

jimjackii, 2025-04-20:
This is here with my ats-mini with the last testversion.

[image]

max-arnold, 2025-04-20:
@Rfromearth Yeah, i suspected that the zero value you got is caused by an older library. You observed the bug described by Dave:

> I tried reading the Display ID (DID) and Display Status (DST) registers with the older TFT_eSPI library that is covered in the discussion thread and noticed that it read back all zeros, so I switched to a more up to date library, which allows the registers to be read.

closed 2025-04-22 by max-arnold