esp32-si4732/ats-mini #46: SSB - AGC gets re-enabled

G8PTN, 2025-04-12:
Hi Max,
I see that someone has questioned why a tone reappears in SSB mode after tuning (noticeable at low volume) when the AGC has been turned off.

The tone is normally present when the AGC is enabled and is a known issue related to the SSB patch. I am not aware of the cause of the tone.

I remembered that that there was an action in the Goshante ATS20_EX code that I commented out, because I didn’t realise why it was required.

It appears that if operating in SSB mode, after an rx.setFrequency(currentFrequency); is issued, for whatever reason the AGC seems to be re-enabled. I don’t think this is documented anywhere.

In SSB mode, this occurs in the functions doFrequencyTuneSSB() and clampSSBBand();

I think the fix is to add doAgc(0); after the rx.setFrequency(currentFrequency); in both of these functions. I did a quick patch in the ATS_MINI code and it appears to fix the issue, but would need further testing.

Function doFrequencyTuneSSB()
[image]

Function clampSSBBand()
[image]

The tone is only really noticeable at low volume or during mute, but is even more apparent when using an amplified speaker.

73, Dave

max-arnold, 2025-04-13:
Thank you! I have seen a similar complaint as well. Will try your recommendation once the refactoring branch is merged

max-arnold, 2025-04-15:
Hi Dave!

I applied the fix, but didn't notice any difference... Could you please try the following build on your side? https://github.com/esp32-si4732/ats-mini/actions/runs/14464142750

max-arnold, 2025-04-15:
There is one report that the patch did fix the issue: https://t.me/talkradio/174172/186346

G8PTN, 2025-04-15:
> I applied the fix, but didn't notice any difference... Could you please try the following build on your side? https://github.com/esp32-si4732/ats-mini/actions/runs/14464142750

Hi Max,
The fix for the AGC being re-enabled appears to work, but whilst testing this, I noticed that the band limits do not work correctly in SSB mode. I assume this bug was introduced during the refactoring.
73, Dave

max-arnold, 2025-04-15:
The fix is available in https://github.com/esp32-si4732/ats-mini/actions/runs/14473452311