esp32-si4732/ats-mini #48: Adding RDS time zone offset to the settings.

Kabuse, 2025-04-16:
It would be cool, if someone could add an offset for the RDS time to macht the local time.
Could be just +/- hours, actual time zones and daylight saving time could be a bit difficult.

Kabuse, 2025-04-19:
I made a fork and implemented the time offset, unfortunately, I messed something up and can not make a pull request:

https://github.com/Kabuse/ats-mini

For the implementation of the "Time Offset" feature, I modified the following files:

Common.h: Here, I declared the global variable timeOffsetHours (extern int8_t timeOffsetHours;).

ats-mini.ino: Here, I defined and initialized the global variable timeOffsetHours (int8_t timeOffsetHours = 0;).

Storage.cpp: Here, I adapted the functions eepromSaveConfig() and eepromLoadConfig() to save and load the timeOffsetHours value in the EEPROM.

Menu.h: Here, I defined a new command CMD_TIMEOFFSET and adjusted the subsequent indices.

Menu.cpp: This involved the most extensive changes:

I added a new "Time Offset" menu item to the settings array.
I adjusted the indices in the settings array and in clickSettings.
I added new functions doTimeOffset() and drawTimeOffset().
I extended doSideBar() and drawSideBar() to call the new functions.
I added #include "Storage.h".
(I may have reordered functions or added forward declarations to resolve compilation errors).

Station.cpp: Here, I modified the showRdsTime() function to apply the timeOffsetHours to the received RDS time before setting the clock.

luarvique, 2025-04-22:
I do not believe these changes are worth including for the following reasons:

1) The receiver has no real RTC and thus forgets time when turned off.
2) RDS information from FM stations often contains incorrect time in random timezones, if any at all.

Hence, obsession with timezones and precise time keeping is somewhat pointless wrt. ATS Mini.

closed 2025-04-22 by luarvique