esp32-si4732/ats-mini #50: V2.10 messed up Weather Fax reception

Kabuse, 2025-04-18:
I can no longer turn to 7880kHz except on band "All" and there, I can't use either "Calibration" or 0.1kHz step, so receiving Weahter-Fax from Hamburg is now impossible for me. The new band list is a huge drawback.

stevecrow74, 2025-04-18:
Indeed, I like that it lists all the ham bands, but there is gaps that I tune to in between, broadcast AM bands and HF air bands and HF marine bands are now harder to get by going through the all option.

max-arnold, 2025-04-19:
Two questions:

* Would a frequency memory feature mitigate the problem for you?
* Please take a look at the current band list https://github.com/esp32-si4732/ats-mini/blob/main/ats-mini/Menu.cpp#L23-L52 Are there any minor adjustments of the band limits that could be done to make you happy?

In the meantime, you can make your own forks and adjust the bands as you want, here is a demo video https://esp32-si4732.github.io/ats-mini/development.html#video-tutorial

max-arnold, 2025-04-19:
> on band "All" and there, I can't use either "Calibration" or 0.1kHz step

You can. Just switch the mode to `LSB` or `USB`

Kabuse, 2025-04-19:
> Two questions:
> 
> * Would a frequency memory or a direct frequency input feature mitigate the problem for you?
> * Please take a look at the current band list https://github.com/esp32-si4732/ats-mini/blob/main/ats-mini/Menu.cpp#L23-L52 Are there any minor adjustments of the band limits that could be done to make you happy?
> 
> In the meantime, you can make your own forks and adjust the bands as you want, here is a demo video https://esp32-si4732.github.io/ats-mini/development.html#video-tutorial

Thank you very much for your prompt feedback and the interesting questions!

Regarding the proposed features, I can confirm that a frequency memory would represent a significant improvement. While navigating through the wide "All" band is currently possible, precisely selecting specific frequencies such as 3855.0 kHz, 7880.0 kHz, and 13882.5 kHz with the small rotary/push encoder can be somewhat cumbersome. A memory for frequently used frequencies would significantly speed up and simplify this process.

I generally find the current band list on Github (https://github.com/esp32-si4732/ats-mini/blob/main/ats-mini/Menu.cpp#L23-L52) to be very well done, especially with regard to the coverage of the amateur radio bands.

Thank you also for pointing out the possibility of creating personal forks and the demo video (https://esp32-si4732.github.io/ats-mini/development.html#video-tutorial). This is very helpful and provides the flexibility to make individual adjustments.

Kabuse, 2025-04-19:
> > on band "All" and there, I can't use either "Calibration" or 0.1kHz step
> 
> You can. Just switch the mode to `LSB` or `USB`

Thank you for pointing that out! Indeed, I overlooked that in my initial assessment. After taking a closer look at the code, the connections are now clearer to me.

I apologize, that was simply my mistake and due to a lack of thoroughness on my part.

max-arnold, 2025-04-19:
> This is very helpful and provides the flexibility to make individual adjustments.

This was one of the goals for this project. It is impossible to fit every possible use-case or feature into one project and keep it well maintained at the same time. Instead, I simplified the build process as much as possible - it is just a single command! And there is even simpler option for small tweaks that requires just a browser and a GitHub account.

So instead of complicating the receiver Menu options or multiplying the amount of Settings, you can have unlimited customization power by changing the code.

Stuff like custom bands or color themes doesn't require any programming knowledge - you just look at the existing data structures and modify them or add new ones.

max-arnold, 2025-04-19:
@stevecrow74 I'm open to expanding the existing band limits a bit (+/- 100-200kHz maybe) to cover some popular out-of-band frequencies.

I just want someone to do the research and make specific suggestions. E.g. the above 3855 kHz can be fitted into the 75M band, etc

We can also consider adding more bands, but the list is already large and scrolling through it is a problem.

stevecrow74, 2025-04-19:
I've had a thought but it would involve a bit of work, and might not be possible, 

But would having a sub menu for Ham, AM broadcast, HF Air and HF Marine be doable?

max-arnold, 2025-04-19:
> But would having a sub menu for Ham, AM broadcast, HF Air and HF Marine be doable

This was suggested on the Telegram chat as well, and it is doable, but I do not like the idea of too many nested menus. Especially because there is no back button. Also adding the sub menus (like Settings) is not very straightforward.

max-arnold, 2025-04-19:
I believe that the memory function (once it is implemented) can serve as a good entry point to various ranges within the ALL band

stevecrow74, 2025-04-20:
I've done a fork with increased bands covering all frequencies, haven't tested it yet. Might slim it down a bit in time. But a start in coding for me anyway. I still have plenty to figure out, and might look into implementing sub menu for the bands.

stevecrow74, 2025-04-20:
I've  released a compiled.zip with All frequencies covered, LW, MW, SW, Air, Ham, from 150Khz to 30Mhz, I've removed Removed All from the band list as everything  is covered in all the bands.

crazy4volvo, 2025-04-23:
Steve, it's the most complete list I've seen in ats-mini forks. But I see one disadvantage - constant loading and unloading of SSB patch. Maybe it would be better to group bands to keep ham, broadcast, air etc. together? I know it will break the ordering by frequency.

stevecrow74, 2025-04-23:
At some stage I plan to have band submenu's for such a thing, all will be what is there now and then broken into SW, Ham, Air, CB, FM, etc etc, with addition of Marine HF channels. 
But like most I'm only just learning C and this is really my first project to learn it with.

luarvique, 2025-04-23:
I have extended several SWL bands to address this problem, at least for now:
```
  {"25M",  SW_BAND_TYPE, AM,  11000, 13000, 11850, 1, 4, 0},
  {"31M",  SW_BAND_TYPE, AM,   9000, 11000,  9650, 1, 4, 0},
  {"41M",  SW_BAND_TYPE, AM,   7000,  9000,  7300, 1, 4, 0},
  {"49M",  SW_BAND_TYPE, AM,   5000,  7000,  6000, 1, 4, 0},
  {"60M",  SW_BAND_TYPE, AM,   4000,  5100,  4950, 1, 4, 0},
```
These changes will be out with the next firmware release.

closed 2025-04-23 by luarvique