esp32-si4732/ats-mini #51: CB Channel Names no longer displayed in V2.10?

SchnauzerLogic, 2025-04-19:
Still new/Still learning so apologies in advance!
I understand that the V2.10 release has:

 "New bands list: more bands, different names and limits, sorted by modulation type for faster switching. WARNING: this change resets the receiver settings!"

Scrolling through the CB channels in the US and don't see any channel names displayed. Wondering if that has been deprecated and if so is there a place in the docs/source code where I might see that so I can answer the easy questions for myself! ;)

Thanks

max-arnold, 2025-04-19:
Good catch, will get it fixed, thanks!

max-arnold, 2025-04-19:
Also this one: https://github.com/esp32-si4732/ats-mini/pull/21#issuecomment-2804425501

luarvique, 2025-04-22:
I believe this has been fixed now.

closed 2025-04-22 by luarvique

max-arnold, 2025-04-22:
The fix was released in 2.11 https://github.com/esp32-si4732/ats-mini/releases/tag/v2.11