esp32-si4732/ats-mini #53: frequency calibration initializtion problem.

cdtsuipo, 2025-04-21:
When I use the AM channel carrier sound for frequency calibration, I have occasionally found that when not switching the band mode from SSB to AM, the calibration frequency data is not automatically loaded when re-entering the band, even though the band is in SSB mode.The regular HAM band could not be tested due to the lack of a continuous signal and the situation is unknown.So, is this ‘bug’ only present in the AM band or in both AM and HAM bands.
Thank you.

max-arnold, 2025-04-21:
Thanks for the bug report!  Does it happen on v1.09?

max-arnold, 2025-04-21:
I reproduced the problem. Calibration is not applied when switching bands or switching the mode to USB/LSB. It is a >=2.0 thing

cdtsuipo, 2025-04-21:
> I reproduced the problem. Calibration is not applied when switching bands or switching the mode to USB/LSB. It is a >=2.0 thing

I tested again, on 30M HAM BAND, 10MHz BPM signal, and the calibration data still didn't apply.

max-arnold, 2025-04-22:
Could you please try the following build to see if it fixes the problem? https://github.com/esp32-si4732/ats-mini/actions/runs/14585127607

cdtsuipo, 2025-04-22:
> > Could you please try the following build to see if it fixes the problem? https://github.com/esp32-si4732/ats-mini/actions/runs/14585127607
> 
> it's works, the same use of AM carrier calibration approach, calibration value has become 1/2 of the previous, before the need to adjust to 1230Hz, now adjusted to 620Hz can not hear the carrier sound, switching bands can also be loaded correctly!
> 
> Thanks!

Correction: the calibration value didn't change much.It should be caused by the weak signal of the AM station I selected.

Based on this version of firmware, I found that when sleep time is set to 0, after closing the screen by long press encoder, I can't wake up the screen by short press on the encoder. ps: do I need to reopen an issue about this.

max-arnold, 2025-04-22:
> I can't wake up the screen by short press on the encoder.

This is the intended behavior. Long press to put it to sleep, the same long press to wake it up

max-arnold, 2025-04-22:
So, if everything works correctly, can I close this issue?

cdtsuipo, 2025-04-22:
> > I can't wake up the screen by short press on the encoder.
> 
> This is the intended behavior. Long press to put it to sleep, the same long press to wake it up

OK, i see.thanks again.

closed 2025-04-22 by max-arnold

max-arnold, 2025-04-22:
The fix was released in 2.11 https://github.com/esp32-si4732/ats-mini/releases/tag/v2.11