esp32-si4732/ats-mini #60: BFO + Calibration

max-arnold, 2025-04-25:
Something is off here...  Tuning the frequency back and forth produces different combinations of the `currentFrequency` (2nd field) and the `currentBFO` (3rd field) on the forward and backward paths. The fourth field is`bandCal` (1430).
 
```
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,193
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,194
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.92,195
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,196
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,197
211,15000,1000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,198
211,15000,1000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,199
211,15000,2000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,200
211,15000,2000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,201
211,15000,3000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,202
211,15000,3000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,203
211,15000,4000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.94,204
211,15000,4000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,205
211,15000,5000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.94,206
211,15000,5000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,207
211,15000,6000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.92,208
211,15000,6000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,209
211,15000,7000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,210
211,15000,8000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,211
211,15000,8000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,212
211,15000,8000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,213
211,15000,9000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,214
211,15000,9000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.94,215
211,15000,10000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.94,216
211,15000,10000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,217
211,15000,11000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,218
211,15000,11000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,219
211,15000,12000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,220
211,15000,12000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.94,221
211,15000,13000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,222
211,15000,13000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,223
211,15000,14000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,224
211,15000,14000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.92,225
211,15000,14000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.94,226
211,15014,1000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.92,227
211,15014,1000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.92,228
211,15014,1000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,229
211,15014,1000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.94,230
211,15014,2000,1430,ALL,USB,1k,3.0k,0,44,62,1,4.93,231
211,15014,2000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,232
211,15014,1000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,233
211,15014,1000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.92,234
211,15014,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,235
211,15014,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,236
211,15014,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,237
211,15014,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,238
211,15014,-1000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.94,239
211,15014,-1000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,240
211,15014,-2000,1430,ALL,USB,1k,3.0k,0,44,62,1,4.93,241
211,15014,-2000,1430,ALL,USB,1k,3.0k,0,44,62,1,4.93,242
211,15014,-3000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,243
211,15014,-3000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,244
211,15014,-3000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,245
211,15014,-4000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,246
211,15014,-5000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,247
211,15014,-5000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,248
211,15014,-6000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,249
211,15014,-6000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.92,250
211,15014,-7000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,251
211,15014,-7000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,252
211,15014,-8000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,253
211,15014,-8000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,254
211,15014,-9000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,255
211,15014,-10000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.91,0
211,15014,-10000,1430,ALL,USB,1k,3.0k,0,44,62,1,4.93,1
211,15014,-11000,1430,ALL,USB,1k,3.0k,0,44,62,1,4.92,2
211,15014,-11000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.92,3
211,15014,-12000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,4
211,15014,-12000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,5
211,15014,-13000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,6
211,15014,-14000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.92,7
211,15014,-14000,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,8
211,15000,-1000,1430,ALL,USB,1k,3.0k,0,44,72,1,4.88,9
211,15000,-1000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.92,10
211,15000,-1000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.92,11
211,15000,-1000,1430,ALL,USB,1k,3.0k,0,44,64,1,4.92,12
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,13
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,64,1,4.93,14
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,15
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,16
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,17
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,18
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,19
211,15000,0,1430,ALL,USB,1k,3.0k,0,44,63,1,4.93,20
```

To get a log like this, connect to the serial port, and press `t`. Tuning and calibration hotkeys are described here: https://esp32-si4732.github.io/ats-mini/manual.html#serial-interface

G8PTN, 2025-04-25:
Hi Max,
This looks like the correct behaviour for the “chuff” free tuning.

If the currentBFO starts at 0, you can tune +/- 14kHz. If you go above/below this, the currentFrequency is set 14kHz higher or lower and then effectively you can move +/- 14 kHz from that. When this action occurs you will here the normal “chuff” sound.

Remember the BFO value sent to the SI4732 then also adds the calibration offset, which is +/- 2000Hz. The maximum BFO value that the SI4732 supports is +/-16 kHz, hence the +/- 14 kHz limit for the “chuff” free tuning. 

73, Dave

G8PTN, 2025-04-26:
Hi Max,
I am not sure why you are using FM?

It is better just to generate a carrier and then tune for the tone. For USB, the tone will be 1kHz when you are tuned 1kHz below the carrier frequency and for LSB when you are tuned 1kHz above the carrier frequency.

You will also hear a tone on the other side because the SI4732 does not fully suppress the unwanted sideband.

You should only need to measure the offset at one frequency, ideally high frequency and then all others can be calculated. Generally, I don’t think the calibration offset will cover the higher bands, because the reference crystal is too far off frequency (typically).

In your example you have an offset 1430 Hz at 15MHz which is 95ppm. So the required offset at other frequencies would be as shown below.

[image]

It would be better to trim the capacitors on the hardware to get the reference frequency closer and then the calibration could probably be removed. Unfortunately this is not very practical for everyone.

I have noticed on the receivers that I have, there is some sort of frequency stability issue. This is most noticeable on SSB and on the higher frequency bands. Touching the antenna connector (GND) changes the pitch of SSB/CW signals. I am not sure if this is parasitic tuning effect on the 32.768kHz crystal or something else.

73, Dave

G8PTN, 2025-04-26:
Hi Max,
For information, an alternative approach to using the BFO for the calibration offset is just to adjust the dial frequency (display). This allows the full +/- 16kHz to be used for the “chuff” free tuning in SSB, but all the stored frequencies and band limits would then need to be adjusted accordingly, and things start to get a little confusing when switching modes. Theoretically, there would be no upper limit to the supported calibration offset.

It should be noted that If the reference results in an offset that is a long way off frequency, it may also affect AM/FM reception, and this can’t be compensated using the BFO method.

Expecting a receiver like this to show the exact frequency in SSB is a little pointless, since the accuracy and stability of the reference is so variable.

As stated previously, the reference should be adjusted to be close to the correct frequency. The SI473x datasheet recommends the reference should be within +/- 50ppm for the seek/tune to work correctly.

[image]

73, Dave

G8PTN, 2025-04-26:
Hi Max,
A further note for general interest, I recently purchased a Radtel RT-860 (same as iradio UV-88) transceiver for around £35 (UK).

This uses two main radio devices.
- BK4819
As used in many cheap VHF/UHF handhelds including the UV-K5.
Frequency range 18 MHz to 660 MHz and 840 MHz to 1300 MHz
Reception modes AM/FM/SSB (SSB is actually double sideband)

- SI4732
As used on the ATS-MINI and many other cheap HF receivers.

The firmware for the SI4732 side is a little clunky and has a few bugs, but via the CPS it does allow up to 128 memories.
Each memory can store Frequency, Mode, Step, Bandwidth, AGC, BFO and Alias (channel name).

[image]

On the BK4819 side, there is a spectrum display, similar to the UV-K5. This does not support simultaneous spectrum and sound.

Different versions are now being released such as the RT-880 (UV-98) and TK11, with additional functionality and appear to be using multiple instances of the BK4xxx chip to achieve functions such as cross-band repeat, dual receive (TK11) and so on.

It will be interesting to see if open source firmware gets developed for these devices. For use as a general scanner/receiver one of the main missing functions is a rotary encoder for tuning, but they do have keypads for direct frequency entry.

73, Dave

max-arnold, 2026-10-13:
Hi Dave!

As a follow up to your comment https://github.com/esp32-si4732/ats-mini/pull/197#issuecomment-3112912777, what do you think about the following approach that can fix the tuning path dependency:

```
carrierHz = ((currentFrequency + 14000) / 28000) * 28000;
bfoHz = (int32_t)currentFrequency - (int32_t)carrierHz;
```

The `currentFrequency` is in Hz.

G8PTN, 2026-10-13:
> carrierHz = ((currentFrequency + 14000) / 28000) * 28000;
> bfoHz = (int32_t)currentFrequency - (int32_t)carrierHz;

Hi Max,
Sorry, I am not sure what this is trying to achieve or am I missing something?
Please could you explain further with examples.

max-arnold, 2026-10-14:
This is a continuation of my earlier [division-remainder proposal](https://github.com/esp32-si4732/ats-mini/pull/197#issuecomment-3112912777): store the frequency in Hz and derive the receiver frequency and BFO from it.

You pointed out that retuning the receiver causes chuffing. Using a fixed 28 kHz grid, rather than a 1 kHz grid, would retain the ±14 kHz BFO span:

- 15.000 MHz → receiver 15.008 MHz, BFO −8 kHz.
- 15.001 MHz → same receiver frequency, BFO −7 kHz.

The same requested frequency would always produce the same combination, regardless of tuning direction. Calibration is added afterward as before. Retuning happens at fixed grid midpoints; repeatedly crossing one could still cause chuffing.

G8PTN, 2026-10-14:
Hi Max,
Yes, this could work, although the chuffing is potentially mitigated with the experimental patches. I still think the fundamental frequency synthesis error issue remains, in that the expected offset correction (ppm) does not remain constant.
73, Dave

max-arnold, 2026-10-14:
Just tried that and I like how using Hz consistently inside the firmware simplifies tuning, validation, persistence, and frequency matching. Converting only at the hardware boundary removes repeated conversions and mode-dependent arithmetic, making the code easier to follow.

Will push the changes soon.

G8PTN, 2026-10-14:
Hi Max,
Yes, I agree working in Hz makes everything neater and using the 28kHz grid for SSB yields the same combination of SI4732 frequency and BFO, so it is a more deterministic behaviour.

If the maths used for the frequency synthesis could be understood it may be possible to account for it, maybe in future SSB patches.