esp32-si4732/ats-mini #69: CPU sleep mode

lasto-com, 2025-05-03:
In CPU sleep mode, we turn off the device. Then we turn it on. The device comes to life for only 3 seconds and turns off. It does not respond to controls. Probably, this should not be so?

max-arnold, 2025-05-03:
Can't reproduce. Could you please post a video (to youtube, dropbox, etc) with the detailed steps that lead to the problem?

Also:

- Firmware version and date?
- What is the Settings->Sleep value?

lasto-com, 2025-05-03:
Sequence of actions.
There was version 2.12 -> Replaced with 2.13.
All changes:
1. Settings -> Sleep mode -> turned on CPU sleep.
2. Settings ->RDS -> ALL (EU).
3. Settings -> Theme -> Space
4. Press the knob for 2 seconds, after a while turn off the receiver.
5. After turning on it can't start. Need to reset the memory.

Played twice, but if you don't change the theme, there is no bug.
The problem may be due to the "wrong" display. I was unlucky with the displayю

max-arnold, 2025-05-04:
Another community member was unable to reproduce the problem on a receiver with the inverted display (DID 9307).

I'll keep this issue open, maybe someone else will help to nail it down.

closed 2025-08-05 by max-arnold