esp32-si4732/ats-mini #71: Size of menu is too small to read.

Roturbo, 2025-05-04:
Hello this firmware is really better than original and others, the only thing i would like to see "if possible" 
is a larger menu on the moment we press the encoder, without glasses i almost can´t read that small words.

Imagine, we click the encoder and at least the central word on menu gets 2 times larger,
after we select it than it returns to normal size.

There is no problem covering other parts of the display because we are focus on the menu only,
or the menu may replace the frequency place with same size words for better view.

[image]

Something almost like this example, the central word is bigger/bolt  and on each side the next words
start to get smaller, the next more smaller,, and all this scrolling from left to right of the screen
using also the freq place, after finish all come back to normal like we see now.  

But, is just one idea.

Regards

max-arnold, 2025-05-04:
I made an experimental version that zooms the current menu item, could you please try it? https://github.com/esp32-si4732/ats-mini/actions/runs/14819839150 (you need to download the `ats-mini-vXXX` artifact and flash the files as usual).

With that said:

- Please do not create Github issues for feature requests, use Discussions instead (issues are only for bugs or tasks that are  planned to be worked on - I prefer to have an actionable list of issues)
- The receiver display is quite small and to get a really accessible UI, the current design needs to be thrown into the bin and reimagined from scratch. I have no plans to do that, but you are encouraged to create your own custom firmware versions.
- Please let me know whether the above firmware version solves your use-case. I'll need to gather more feedback from other users before releasing it, and it is quite likely that the zoomed text will be disabled by default in the Settings menu, because it covers RDS info and frequency names (CB channels and popular frequencies like SSTV and FT8)

Roturbo, 2025-05-05:
Hello @max-arnold many tanks, it really works, at start i was unable to update the radio because it have gone in to sleep mode,
so i have to reboot and increase the sleep time.

I didn´t think about that option, but it´s fine, i´m able to read it since it is bigger, and i have a friend
that also agree and will update with this version.  

**_Please do not create Github issues for feature requests, use Discussions instead (issues are only for bugs or tasks that are planned to be worked on - I prefer to have an actionable list of issues, not a world-writeable TODO list that constantly grows)_**

Sorry, there are some repositories were we chose from issue or request on same place, next time i will use Discussions. 

**_The receiver display is quite small and to get a really accessible UI, the current design needs to be thrown into the bin and reimagined from scratch. I have no plans to do that, but you are encouraged to create your own custom firmware versions._**

Yes it is small display, but now is more easy to use with your mode, for me i´m not a programmer, i know little things only. 

**_Please let me know whether the above firmware version solves your use-case. I'll need to gather more feedback from other users before releasing it, and it is quite likely that the zoomed text will be disabled by default in the Settings menu, because it covers RDS info and frequency names (CB channels and popular frequencies like SSTV and FT8)_**

Yes, like i say above, i can read it now, this may be very usefully for people with difficult to read small things.

About covering RDS and other things, in my opinion is not a problem, i´m using the menu to change
band, mod or other things and i don´t need to watch that details at the same time because they are
not important on that moment, other radios use the Menu to cover all display, other person may 
have different opinion that´s why we call it "opinion".

In general, your code is the best that i have see and use, many many functions that are important 
to save battery and lots of info on each band, channel etc, it is just amazing. 

The only thing missing now that may not be possible, is a DSP for better audio on SSB.

Once again, many tanks for your work, you may close this "issue" that was a request.

Regards.

max-arnold, 2025-05-05:
Thanks!

The feature will be released in 2.14 this week.

closed 2025-05-05 by max-arnold