esp32-si4732/ats-mini #117: Low speaker volume when turning on the receiver

denka977, 2025-05-25:
On the latest firmware versions (this was not the case on 1.01), when turning on the receiver, for example in CityzenBand, the volume from the speaker is extremely low. But it is worth changing the signal gain parameter (to any value) or switching to any frequency from those saved in memory, or switching to another band - the volume is restored to normal. The problem occurs only when turning on the receiver. The settings do not matter (I tried it after a full reset).

**To Reproduce**

turn on the receiver on any frequency in AM, it is desirable that it turns on itself as the last saved one and you will hear a quiet sound from the speaker. change any settings (the volume parameter affects the sound level only within the already low volume, therefore, to return the volume to a normal level, change other settings).

G8PTN, 2025-05-25:
Hi,
There appears to be an error in the doAvc() translated function in the Menu.cpp file.
This means that the Avc value will not be updated at power-up.

[image]

Change: if(currentCmd==FM) return;
To: if(currentMode==FM) return;

73, Dave

max-arnold, 2025-05-25:
@denka977 Could you please test this build? https://github.com/esp32-si4732/ats-mini/actions/runs/15237492302

denka977, 2025-05-25:
> [@denka977](https://github.com/denka977) Could you please test this build? https://github.com/esp32-si4732/ats-mini/actions/runs/15237492302

according to my tests, the bug is gone
good job!