esp32-si4732/ats-mini #128: ATS Mini Controller

GrzegorzZajac000, 2025-05-28:
Hi!

I've developed a tool for controlling the ATS Mini 
https://ats-mini-controller.vercel.app/

While working on this tool, I've encountered a few issues and thought of some improvements that could make the experience much better. Here are my suggestions and bug reports:

1. Screenshot Speed
Taking screenshots currently takes a very long time. Is there any way to speed up the screenshot process?

2. Audio Quality Issue (USB + 3.5mm Jack)
When the ATS Mini is connected to a PC via both USB and the 3.5mm jack at the same time, the audio output is very weak and heavily distorted (lots of clipping). Is there a way to improve this? A similar situation occurs when connecting to a phone via USB-C and playing audio through the ATS MINI speaker.

3. Saving Theme via USB Serial Command
Would it be possible to add a command to save the current theme through the USB serial interface?

4. Querying Single Values via Command
Could you add a command that allows querying a single value, such as the current mode, band, volume, step, seek, screen on/off, etc.?

5. Setting Values Directly via Command
Would it be possible to add commands to directly set values like seek (e.g., to 100k, 1M, etc.), as well as other parameters such as mode, band, volume, and other settings?

Thanks for your great work on ATS Mini! These changes would make the tool much more powerful and user-friendly.

**To Reproduce**
Go to https://ats-mini-controller.vercel.app/ and connect radio ;)

**Expected behavior**
Screenshots are captured significantly faster.
Audio quality remains clear and strong when both USB and 3.5mm jack are connected.
Theme can be saved via a simple USB serial command.
Single device parameters (mode, band, volume, etc.) can be queried with dedicated commands.
Device parameters (seek, mode, band, volume, etc.) can be set directly via command with specific values.

**Information:**
2.22

max-arnold, 2025-05-29:
- There is nothing I can do about the audio quality (it is a hardware issue)
- Screenshot speed - unlikely. We have very little RAM left for compression, and the baud rate seems to have no effect
- Saving themes is not currently possible (they are effective until restart). This might change in future, but not until #94 is done
- The rest will be handled by the CAT protocol support I'm (slowly) working on. I'm also considering an extended command set.

In future, please use issues only for bug reports. Feature suggestions should go into Discussions.  But I'll leave this issue open since I'm already working on CAT.

max-arnold, 2025-09-04:
@RadioHam25 Please use discussions to ask questions (this github issue is about remote control).

We do not support Arduino IDE, but you can find some hints here https://github.com/esp32-si4732/ats-mini/discussions/148

max-arnold, 2026-11-14:
Related https://github.com/esp32-si4732/ats-mini/discussions/404