esp32-si4732/ats-mini #201: On the CB band, the volume is very, very low on AM.

maxlinux2000, 2025-07-26:
**Describe the bug**
On the CB band, the volume is very, very low on AM.

**To Reproduce**
a) On the CB radio, set channel 9 to AM, for example.
b) On the ATS, set the CB band and select the frequency 27065 kHz AM - 6 kHz bandwidth.
c) Start transmitting on the CB radio.

Result: You can barely hear anything on the mini radio, and the SMeter barely moves.

**Expected behavior**
You should hear AM transmissions loud and clear on 27 MHz.

**Information:**
Firmware 1.01, but the same thing happens with the latest 2.13.

I'm thinking of recommending this radio for a 72-hour emergency backpack.
It would be ideal because it's very small, rechargeable, receives commercial AM and FM radio, but also allows you to listen to amateur radio emergency frequencies.

In my area, channel 9 AM on the CB band is widely used, which is why I'm interested. The problem is that I can barely hear anything.

Regards,
MaX

max-arnold, 2025-07-27:
Hey Max!

I'm not sure what to do with this issue. You are first who reported this problem, and I neither have a CB transceiver nor can receive many AM signals on CB to test that myself.

Software-wise there is no difference between AM on the broadcast bands and on CB. Si47xx Programming Guide also doesn't seem to have many tunables for AM (besides AGC/AVC that are already exposed in the receiver's Menu options).

Maybe you can try a resonant CB antenna?

max-arnold, 2025-08-05:
There is nothing I can do to fix this issue