esp32-si4732/ats-mini #215: Wifi connection

maurilored-stack, 2025-08-14:
With version 2.30, I'm having WiFi issues. The radio often won't connect. Under the same conditions as with 2.28, this didn't happen.

maurilored-stack, 2025-08-15:
> I just switched my receiver on and off 10 times in a row (in Sync Only
> mode) and unfortunately can't reproduce the problem. Let's wait for similar
> complaints from other people.

Many Thanks for your answer.
It almost never works in sync-only mode. Sometimes it works when
connection or connection  + AP, but it takes a few minutes to connect,
and not always.
It seems slow in negotiating the router's response. With 2.28, it was
lightning fast!

max-arnold, 2025-08-15:
Did you flash the right version (ospi/qspi, so the PSRAM amount is non zero)?
Have you tried flashing a merged file instead of three separate ones?
What happens when you roll back to 2.28?

DwaineGarden, 2025-08-27:
I can confirm I don't have this issue.   Never experienced any disconnect during the loading of

max-arnold, 2025-09-15:
@maurilored-stack How many access points you have configured in the ATS Mini web UI? Just one or 2-3?

max-arnold, 2025-09-17:
Closing this since you confirmed that WiFi works in 2.32

closed 2025-09-17 by max-arnold