esp32-si4732/ats-mini #219: Click/thump now present on V3 receiver with version 2.28 firmware and above?

SuspectTheGoose, 2025-08-20:
Hello,

I have a version V3 receiver and I noticed that when upgrading from version 2.27 firmware to 2.28 or above there is a noticeable thump/click when changing bands or when cycling through memories in the memory bank. Neither of these were present on version 2.27.

I see that in the changelog for 2.28 this is shown:

"Fix loud clicks when changing bands/modes on the PCB version without the mute circuit. [#103](https://github.com/esp32-si4732/ats-mini/issues/103)"

Could this have perhaps caused an issue with V3 receivers?

max-arnold, 2026-10-17:
Could you please test the following build https://github.com/esp32-si4732/ats-mini/issues/385#issuecomment-6230884919

It might fix the click/thump issue on V3/V4

mickecamino, 2026-10-17:
OK, I just tested the build. No thump when going through the bands. No thump when start Scan ocr when the scan is completed.
I think you nailed it 👍

max-arnold, 2026-10-17:
Mikael, thank you for confirming that!

The fix is going to be released in v2.44