esp32-si4732/ats-mini #229: Calibration in memories

maurilored-stack, 2025-09-03:
Hi, I use version 2.30 and I'm having this problem.
If I listen to frequencies on different bands, such as 13.250, 7100, and 3560, both in USB and LSB, which require different calibrations, when I save them to memory, the individual calibrations are not retained, and I have to retrain each frequency. Is there a way to avoid this?
Thanks

max-arnold, 2025-09-04:
Calibrations are saved per-band, each memory slot holds a reference to the corresponding band.

In the upcoming 2.31 each band will have independent calibration values for LSB/USB. See https://github.com/esp32-si4732/ats-mini/pull/220 for more details

max-arnold, 2025-09-04:
You can try the following test build https://github.com/esp32-si4732/ats-mini/actions/runs/17411993916

maurilored-stack, 2025-09-04:
Many thanks, good work with this little radio!!!