esp32-si4732/ats-mini #244: Cannot connect to the WiFi network in the configuration slot 3

yarda, 2025-09-14:
**Describe the bug**
If I put my AP configuration to the Network slot 3 it doesn't connect (the previous 2 slots have configured networks). If I put it to the slot 1, it connects OK.

**To Reproduce**
Always with my APs

**Expected behavior**
No difference in ability to connect to the WiFi no matter in which slot the WiFi network configuration is (maybe higher connection delay for the networks in the higher slots).

**Information:**
 - v2.31

**Additional context**
This has simple workaround, but it would be nice to have it correctly working. Maybe just increasing the connection timeout may work (my networks in the slots 1 and 2 aren't available).

paolettopn, 2025-09-15:
Hi Max,
I installed this 'fixed' version normally, and when I turned on the radio, it connected normally via WiFi on any of the three WiFi network configurations.
All this WITHOUT having to reset the radio settings after installing the firmware.
I think we're done!

max-arnold, 2025-09-16:
The WiFi fix was released in 2.32.

max-arnold, 2025-09-17:
@1956jurikh See https://github.com/esp32-si4732/ats-mini/issues/247